JTrotta/RaspberrySharp #1: Scan I2C Bus

moelski, 2018-03-16:
Hi !

Just stumbled about this new lib. I´m amazed :-) Great work !!

Is it possible to make a I2C scan with this lib? I couldn´t find any functions for that. 
It it possible in general ? 

Dominik

JTrotta, 2018-03-16:
Hi!,
yes it's possible. I created a milestone for first production release of the project.
JJ

moelski, 2018-03-17:
Great. 
If you have any testversion ... let me know :-)

moelski, 2018-04-14:
Did u make any progress on this topic ?

moelski, 2018-04-14:
Will do so if it ´s checked in :-)

JTrotta, 2018-04-14:
hooooooops...

moelski, 2018-04-17:
So my last PR for detials ...

JTrotta, 2018-04-19:
Updated. Have  check.

moelski, 2018-04-20:
Hi !
Did not work. The I2CDetect Class did not work: 
```
Unhandled Exception:
System.InvalidOperationException: Read operation failed with BCM2835_I2C_REASON_ERROR_NACK status
  at RaspberrySharp.IO.InterIntegratedCircuit.I2cDriver.Read (System.Int32 deviceAddress, System.Int32 byteCount) [0x00129] in <a48c02ccd90a46e6bcfacd23f78faf73>:0
  at RaspberrySharp.IO.InterIntegratedCircuit.I2cDeviceConnection.Read (System.Int32 byteCount) [0x00001] in <a48c02ccd90a46e6bcfacd23f78faf73>:0
  at RaspberrySharp.IO.InterIntegratedCircuit.I2cDetect.Detect () [0x0001d] in <a48c02ccd90a46e6bcfacd23f78faf73>:0
  at I2CDetect.Program.Main (System.String[] args) [0x00021] in <c432a5608ce5456abc971708b48e197f>:0
[ERROR] FATAL UNHANDLED EXCEPTION: System.InvalidOperationException: Read operation failed with BCM2835_I2C_REASON_ERROR_NACK status
  at RaspberrySharp.IO.InterIntegratedCircuit.I2cDriver.Read (System.Int32 deviceAddress, System.Int32 byteCount) [0x00129] in <a48c02ccd90a46e6bcfacd23f78faf73>:0
  at RaspberrySharp.IO.InterIntegratedCircuit.I2cDeviceConnection.Read (System.Int32 byteCount) [0x00001] in <a48c02ccd90a46e6bcfacd23f78faf73>:0
  at RaspberrySharp.IO.InterIntegratedCircuit.I2cDetect.Detect () [0x0001d] in <a48c02ccd90a46e6bcfacd23f78faf73>:0
  at I2CDetect.Program.Main (System.String[] args) [0x00021] in <c432a5608ce5456abc971708b48e197f>:0
```

JTrotta, 2018-04-20:
With last commit it should work.
I'm going to close this issue, if you got any problem reopen it.

closed 2018-04-20 by JTrotta