JTrotta/RaspberrySharp #2: How to implement new samples?

moelski, 2018-03-17:
Hi !
Since I would like to support this project I have a general question. 
What´s the best way to create / add new demos for other Sensors / Inputs? 
In the "old" RaspberrySharop.IO days there where connector classes. 
In you actual demo folder you have standalone applications for sensors. 

What´s the way you prefer? 

regards
  Dominik

JTrotta, 2018-03-18:
What you mean for "there where connector classes".
May you show me any example? If it was a good (and modern) system we could use it again

moelski, 2018-03-19:
Hi !
At the old project they had a special namespace colled "Raspberry.IO.Components". 
Take a look here: https://github.com/raspberry-sharp/raspberry-sharp-io/tree/master/Raspberry.IO.Components
It is reusable which is fine. But it used external libs like unitsnet. 

In general it would be nice to have such a namespace for components. Combined with a sample to each component and maybe some fritzing wiring instructions. That way it would be much easier to use for others. 

What´s your opinion using the unitsnet lib? 
-> https://github.com/angularsen/UnitsNet

Dominik

JTrotta, 2018-04-07:
You did it

closed 2018-04-07 by JTrotta