JTrotta/RaspberrySharp #12: Generating PWM Signals

moelski, 2018-04-27:
Hi !
There is one thing missing in this Lib ... generating PWM signals. 
At the Moment I see 3 different ways .... 

1) Using the Hardware like in this lib: http://www.ofitselfso.com/RPICSIO/Help/RPICSIOHelp_PWMPortExample.html
```
    # enable one PWM with GPIO_18, header Pin12 by default
    dtoverlay=pwm.dtbo
    # enable two PWMs GPIO_18, header Pin12 and GPIO_19, header Pin35 by default
    dtoverlay=pwm-2chan.dtbo
```
2) Using Software PWM with Pigpiod as a service (can be accessed by TCP). I used that in some cases and it works really well. 
3) Writing an own Software PWM class - but at the moment I have no idea how fast we can toggle Pins...

What do you think about generating PWM directly on the Pi ?

Dominik

JTrotta, 2018-04-27:
We can try.
But I remember on old P1, PWM was very CPU expensive. 
Now I'm using external chips, but we can try on new pis.
I do not have any preference; point 1 should be tested, point 2 do not know you can try , point 3 should be easy, we can toggle at 1ms.

JTrotta, 2018-05-01:
Good news,
I going to implement the original bcm2835  PWM code. I will create a new folder like InterIntegratedCircuit that will contain all PWM driver code.
Stay tuned.

moelski, 2018-05-02:
Hmm that means we will only have 2 Channels, correct ? 
I will place my pigpio based code the next days as a PR. 
Not the best solution, but itś ok for controlling Servos and Motors. 

Dominik

JTrotta, 2018-05-02:
Yes 2 channels gpio 18 and 19.
If we want more, we need external chip like TLC5940  or PCA9633.

wezzix, 2018-10-16:
I was just looking for PWM functionality. Did you make it work?

JTrotta, 2018-10-18:
@wezzix I did not do any test on PWM.