JTrotta/RaspberrySharp #19: OneWire support

Samurai2083, 2018-12-12:
Hi, I was wondering if you are planning to get a running sample for the OneWire support? I'm in the middle of trying to work out building a C# project for Raspberry Pi Zero, and instead of using Windows IoT Core (which is not supported on Raspberry Pi Zero), I'm looking into porting my project to run on Raspbian and mono runtime. 

The sensors I'm currently trying to run from Raspberry Pi Zero are:

DS18B20 (temp sensor)
TSL2561 (Light sensor)
MCP3008 (to connect to a moisture sensor)

I have these running on the Raspberry Pi 3, but I would like to reduce the footprint of the device(s), and the Raspberry Pi Zero seems to be a good choice. I usually do all my work in Visual Studio building apps in C#.

With that said, I've been looking into your library and would be interested in using it if I could use it to support those devices I listed. 

I understand it's a work in progress, and if you have any suggestions for me, or thoughts on if your library could support those devices currently, please let me know.

JTrotta, 2018-12-13:
Hello, Yes I have some demos using OneWire. I will add them ASAP.
The library is under test but I'm using it in production with no problems. In few weeks, I will update nuget to production.

Samurai2083, 2018-12-14:
>           Hello, Yes I have some demos using OneWire. I will add them ASAP.
> The library is under test but I'm using it in production with no problems. In few weeks, I will update nuget to production.

Hey, thank you. Much appreciated. I'll check back from time to time to see what is in the repo and give it a try. Thank you again!

JTrotta, 2020-03-21:
Onewire works now

closed 2020-03-21 by JTrotta