JTrotta/RaspberrySharp #21: .Net 2.0 targeting

pmeloy, 2019-01-23:
When I load the project in VS2017 I get an error saying:

Severity	Code	Description	Project	File	Line	Suppression State
Error		The current .NET SDK does not support targeting .NET Standard 2.0.  Either target .NET Standard 1.6 or lower, or use a version of the .NET SDK that supports .NET Standard 2.0.	RaspberrySharp	D:\Program Files (x86)\Microsoft Visual Studio\2017\Community\MSBuild\Sdks\Microsoft.NET.Sdk\build\Microsoft.NET.TargetFrameworkInference.targets	126

I've downloaded and installed every SDK listed by the projects from 4.6.1 to 4.7.2. What am I missing?

JTrotta, 2019-01-23:
You have to use .NET 4.6.2 or more, or .NetStandard 2.0

pmeloy, 2019-01-23:
Pulled from github yesterday and it won't even load without 4.6.1, 4.6.2, 4.7.1, 4.7.2 and .NET Core 2.

Each project seems to have different requirements and won't play with higher versions.

Still, I managed to make everything happy write a components for the BMP180 and I'm working on ADC1015.

JTrotta, 2019-01-23:
Strange, tomorrow will check.

pmeloy, 2019-01-24:
Maybe it's my VS2017 settings? I get lots of reports of "e" not being declared but never used. I have that reporting turned off in VS2015 and haven't really used 2017 until now.