JTrotta/RaspberrySharp #24: MemoryMapFailedException at Blinkexample

aBinaryStory, 2019-02-24:
Hello, i was testing your SampleProject "GPIO.Blink" using a RPI 3 Model B 3+ and the repository Version 1.3.2 and once 1.3.3 Beta from Nuget.Org. 
Following Error occurs when the program reaches this line:

`conPin = new GpioConnection(ProcessorPin.Gpio16.Output());`

Unhandled Exception:
RaspberrySharp.IO.Interop.MemoryMapFailedException: Exception of type 'RaspberrySharp.IO.Interop.MemoryMapFailedException' was thrown.
  at RaspberrySharp.IO.Interop.MemoryMap.ThrowOnError[TException] (System.IntPtr result) [0x00016] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.Interop.MemoryMap.Create (System.IntPtr address, System.UInt32 size, RaspberrySharp.IO.Interop.MemoryProtection protection, RaspberrySharp.IO.Interop.MemoryFlags memoryflags, System.Int32 fileDescriptor, System.UInt32 offset) [0x00019] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnectionDriver..ctor () [0x00051] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnectionSettings.GetBestDriver (RaspberrySharp.IO.GeneralPurpose.GpioConnectionDriverCapabilities capabilities) [0x0000f] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnectionSettings.get_DefaultDriver () [0x00010] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnectionSettings..ctor () [0x00008] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnection..ctor (RaspberrySharp.IO.GeneralPurpose.GpioConnectionSettings settings, System.Collections.Generic.IEnumerable`1[T] pins) [0x0002e] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnection..ctor (RaspberrySharp.IO.GeneralPurpose.PinConfiguration[] pins) [0x00000] in <807f0c5432f14949b24392f867d967bd>:0
  at Blink.Program.Main () [0x00030] in <efb2d7af641f4c6685cfeead2cd2653c>:0
[ERROR] FATAL UNHANDLED EXCEPTION: RaspberrySharp.IO.Interop.MemoryMapFailedException: Exception of type 'RaspberrySharp.IO.Interop.MemoryMapFailedException' was thrown.
  at RaspberrySharp.IO.Interop.MemoryMap.ThrowOnError[TException] (System.IntPtr result) [0x00016] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.Interop.MemoryMap.Create (System.IntPtr address, System.UInt32 size, RaspberrySharp.IO.Interop.MemoryProtection protection, RaspberrySharp.IO.Interop.MemoryFlags memoryflags, System.Int32 fileDescriptor, System.UInt32 offset) [0x00019] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnectionDriver..ctor () [0x00051] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnectionSettings.GetBestDriver (RaspberrySharp.IO.GeneralPurpose.GpioConnectionDriverCapabilities capabilities) [0x0000f] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnectionSettings.get_DefaultDriver () [0x00010] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnectionSettings..ctor () [0x00008] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnection..ctor (RaspberrySharp.IO.GeneralPurpose.GpioConnectionSettings settings, System.Collections.Generic.IEnumerable`1[T] pins) [0x0002e] in <807f0c5432f14949b24392f867d967bd>:0
  at RaspberrySharp.IO.GeneralPurpose.GpioConnection..ctor (RaspberrySharp.IO.GeneralPurpose.PinConfiguration[] pins) [0x00000] in <807f0c5432f14949b24392f867d967bd>:0
  at Blink.Program.Main () [0x00030] in <efb2d7af641f4c6685cfeead2cd2653c>:0

Any idea's how i can solve this ?

tylerforsythe, 2019-02-24:
Are you running it with escalated privileges, e.g. sudo mono testapp.exe? I've found it has to be run with sudo to interact with GPIO.

JTrotta, 2019-02-25:
Yes please. First try: sudo mono appname.exe.
Then, let us know if it works (it should).

aBinaryStory, 2019-02-25:
You are right,
"sudo mono appname.exe." solved the problem!
Thank you so much for keeping the Raspberry Repository for new Models alive, this is amazing!

closed 2019-02-25 by aBinaryStory