JTrotta/RaspberrySharp #25: How do I activate the pull-down resistor for a pin?

Icezman001, 2019-06-19:
I'm having issue with getting random values (true/false) when reading a pin (checking if enabled or not).
Only when I push the button, it steadily gives false but all the other times it's just unreliable.

I found this:
https://raspberrypi.stackexchange.com/questions/63189/activating-pull-down-resistors-in-c

They say i should work with the pulldown but I can't seem to find it anywhere or atleast no idea how to affect it in any way.

Could someone help?

tylerforsythe, 2019-06-20:
I've never done pull-down stuff before, but I think some code to get you started is:

`var pinConfig = new InputPinConfiguration(ProcessorPin.Gpio0);`
`pinConfig.Resistor = PinResistor.PullDown;`
`var myGpioPin = new GpioConnection(pinConfig);`

Please report back if that works or not. Thanks!

Icezman001, 2019-06-22:
> I've never done pull-down stuff before, but I think some code to get you started is:
> 
> `var pinConfig = new InputPinConfiguration(ProcessorPin.Gpio0);`
> `pinConfig.Resistor = PinResistor.PullDown;`
> `var myGpioPin = new GpioConnection(pinConfig);`
> 
> Please report back if that works or not. Thanks!

Thanks for the input! It really helped!
I finally got it working with this:
```
conPinButton = new GpioConnection(ProcessorPin.Gpio16.Input().PullUp());
```
Probably not the best way haha, so your solution is better I think!
Here's my full code (not optimized or anything):
```
using RaspberrySharp.IO.GeneralPurpose;
using System;
using System.Linq;
using System.Threading;

namespace BreadBoard
{
    class Watchpin
    {

        private static GpioConnection conPinButton;
        private static GpioConnection conPinLed;

        private static bool ledActivated;
        private static bool buttonPressed;

        static void Main(string[] args)
        {
            conPinButton = new GpioConnection(ProcessorPin.Gpio16.Input().PullUp());
            conPinLed = new GpioConnection(ProcessorPin.Gpio24.Output());
            

            Console.WriteLine("GPIO: " + conPinButton.Pins.First().Configuration.Pin);
            conPinButton.PinStatusChanged += ConPin_PinStatusChanged;
            buttonPressed = conPinButton.Pins.First().Enabled;
            ledActivated = conPinLed.Pins.First().Enabled;

            while (true)
            {
                if (Console.KeyAvailable)
                {
                    ConsoleKeyInfo key = Console.ReadKey(true);
                    switch (key.Key)
                    {
                        case ConsoleKey.X:
                            conPinLed.Pins.First().Enabled = false;
                            Console.WriteLine("Program End ...");
                            Thread.Sleep(1000);
                            Environment.Exit(0);
                            break;
                    }
                }

                conPinLed.Pins.First().Enabled = false;
                //if (!conPinButton.Pins.First().Enabled)
                //{
                //    Console.WriteLine("Button pressed: " + buttonPressed);
                //    conPinLed.Pins.First().Enabled = true;
                //}
                //else
                //{
                //    //Console.WriteLine("Button not pressed!");
                //    conPinLed.Pins.First().Enabled = false;
                //}
                Thread.Sleep(50);
            }
        }

        private static void ConPin_PinStatusChanged(object sender, PinStatusEventArgs e)
        {
            if (e.Enabled)
            {
                Console.WriteLine("Button PRESSED!");
                conPinLed.Pins.First().Enabled = true;
                Thread.Sleep(1000);
            }
        }
    }
}
```
Hope it helps people!

JTrotta, 2019-06-22:
@Icezman001 Take  a look at /samples/monitors/gpiomonitor
It may help you.

closed 2019-06-22 by JTrotta