JTrotta/RaspberrySharp #26: Connection to Adafruit i2c 16x2 RGB LCD Pi Plate

kentew, 2020-01-03:
I am trying to use this to connect to the plate mentioned in subject. I am using C# in Visual Studio 2017, then running on the RaspberryPi using Mono. I have had success with multiple applications this way, including one that connects to a LED light. I am just not having any luck with the 16x2 character LCD. I'll try to organize my efforts again and get an error message to you, but really just looking for some general direction as I am completely frazzled right now with all the various attempts.
Thanks,
Ken

bfsmithATL, 2020-01-03:
Hey Ken, 
I’d first try to see if your LCD Pi plate is OK. Do you have another I2C host like an Arduino or other microcontroller handy to send data to the plate? Have you been able to verify that your Pi is sending the proper I2C frames?  You can use a logic analyzer like the [Saleae Logic](https://www.saleae.com/) or equivalent, very handy.  Also, besides peppering your code with logging and console out statements to debug along the way, try setting up remote debugging from Visual Studio to mono on the api, very very useful: check [Here](https://marketplace.visualstudio.com/items?itemName=Bongho.MonoRemoteDebugger) and this [Really good write up ](https://www.hanselman.com/blog/RemoteDebuggingWithVSCodeOnWindowsToARaspberryPiUsingNETCoreOnARM.aspx)

kentew, 2020-01-03:
Thanks for the quick response. I have determined the board/display works
fine using python on the same raspberry pi. And, the biggest help, is that
in all the confusion, it looks like I was running a mish-mash of this
RaspberrySharp and the previous version. I think I am on the right track
now as I have gotten an I2Cdriver to create. One other quick question,
while browsing the code, I noticed a lot of empty directories. Is that just
because that code has been ported?

Thanks,
Ken

JTrotta, 2020-01-04:
@kentew Which empty directories are you talking about?

kentew, 2020-01-04:
Under components for example. I am working with some Displays and I don't
see any code in there when I open the project. There was in the original
Raspberry project.

JTrotta, 2020-01-04:
Ok. No it's not because it has been ported, it's just because I published only what I have already tested. I did not ever used any display. When I'll use them I will publish the code.
Any way you are free to pull request, if you have any working code.