JTrotta/RaspberrySharp #27: New models?

iminet, 2020-04-09:
Hey, what's about new models, such as Pi 4 variants?
This lib is great but I'm really missing the new models and new features as well.
Is this project also abandoned like it's predecessor?

JTrotta, 2020-04-09:
No the project is still alive. I'm working to move some features to the new [system.device.iot](https://github.com/dotnet/iot).
When the Microsoft project will be ready, than this one will die.
I do not have a PI4, but I can add it if you need.

iminet, 2020-04-09:
Sure, it would be great, thank you! I tried to use some other information what can be found in the files /proc/cpuinfo and /proc/meminfo. Let me know, if you need these snippets.

JTrotta, 2020-04-09:
Chcek if the result is like this, or, better, copy and paste me the result you get:

processor : 0
model name : ARMv7 Processor rev 3 (v7l)
BogoMIPS : 108.00
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd08
CPU revision : 3

processor : 1
model name : ARMv7 Processor rev 3 (v7l)
BogoMIPS : 108.00
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd08
CPU revision : 3

processor : 2
model name : ARMv7 Processor rev 3 (v7l)
BogoMIPS : 108.00
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd08
CPU revision : 3

processor : 3
model name : ARMv7 Processor rev 3 (v7l)
BogoMIPS : 108.00
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd08
CPU revision : 3

Hardware : BCM2835
Revision : c03111

CsharpAndVB, 2020-08-27:
Hi there,
Can this lib. be used on a Pi 4b?
If not then could you please let me know how else I can access Pi 4b peripherals & gpio using c# / mono.
Thank you!

JTrotta, 2020-08-31:
@CsharpAndVB  Yes.

CsharpAndVB, 2020-08-31:
Super. Thank you!

CsharpAndVB, 2020-08-31:
Hi,
I just compiled the Blink example in VS2019 community, copied over Blink.exe and RaspberrySharp.dll to the Pi 4b and tried to run it.
Unfortunately it just crashes and the DOS window disappears after half a second or so.
The initial connection line causes the crash:

**conPin = new GpioConnection(ProcessorPin.Gpio16.Output());**

If I remove all references to the GPIO then the program works (obviously).

Any help getting this working would be most appreciated.

Thank you

tylerforsythe, 2020-08-31:
> Hi,
> I just compiled the Blink example in VS2019 community, copied over Blink.exe and RaspberrySharp.dll to the Pi 4b and tried to run it.
> Unfortunately it just crashes and the DOS window disappears [...]

Did you run as sudo? That’s the issue 99% of the time.

CsharpAndVB, 2020-08-31:
> Hi,
> I just compiled the Blink example in VS2019 community, copied over Blink.exe and RaspberrySharp.dll to the Pi 4b and tried to run it.
> Unfortunately it just crashes and the DOS window disappears [...]

Hi,

Sorry, I'm new to the Pi therefore I don't know how to do that.
I've tried "sudo Blink.exe" in the Pi terminal but this doesn't work?

How do I do it?

Thank you!

CsharpAndVB, 2020-08-31:
@JTrotta 
Hi, I just tried the demo. app. The log file also reports the same error "Object ref. not set to an instance of an object".
Thanks

CsharpAndVB, 2020-09-01:
@JTrotta 
Hi, sorry to trouble you but could you please help me get this working on a 4b?

I'm just trying to get the BLINK example working and it appears the DRIVER object is NULL.

Thank you

Graveen, 2020-09-02:
you tryed "sudo mono blink.exe" ? Unfortunatly i does not have 4b so i can't really troubleshoot
++

CsharpAndVB, 2020-09-02:
@Graveen 
Hi, yes - I've tried that and I get the same "Object ref. not set to an instance of an object." error.

If you run the Blink program on your windows machine you will see the same error when the program tries to create a new IO connection.

Thanks for your help!

CsharpAndVB, 2020-09-02:
I think the problem is that the software actually isn't compatible with the 4b. I notice there's a GetModel function and the 4b is not listed in the Switch statement, so "unknown" is returned.
@JTrotta 
Can you please add the 4b to this?
Thank you

JTrotta, 2020-09-03:
Hello,
I do not have a 4b yet. As soon as I got a 4b, I will add it to RaspberrySharp.

CsharpAndVB, 2020-09-03:
@JTrotta 
Thanks, but that could be months from now & I was hoping to get this working soon.
I will try to get it working myself but I can't get remote debugging to work either so it's all a bit of a pain! :-)